space-concordia-robotics/robotics-prototype #275: Setup TX2 initialization

PeterGhimself, 2020-01-12:
We are switching to the Jetson TX2 as our new OBC away from the Odroid XU4.

We know that through @KoaTheKoala 's efforts we have ubuntu 16.04 and ros kinetic installed on our TX2. Now we must setup the repo and systemd service to ensure that on startup everything will behave the same way as before. There are probably going to be a number of breaking changes that need to be dealt with.

NinjaWeeaboo, 2020-02-24:
I started some work on this recently, I changed some directory constants in the scripts and systemd services, in order for them to work on the TX2. So because there are code changes, I'm going to make a new branch and commit them when I get a chance.

vashmata, 2020-03-11:
idk if it applies to this issue but some things are apparently sensitive info that we don't want other people to see so just be aware of that when doing this type of stuff. I'm not sure which files or which information we want to keep hidden so it may not apply here

NinjaWeeaboo, 2020-03-14:
> idk if it applies to this issue but some things are apparently sensitive info that we don't want other people to see so just be aware of that when doing this type of stuff. I'm not sure which files or which information we want to keep hidden so it may not apply here

I think this must just apply to the stuff under the elec repo on Gitlab. And I believe the code in their repo just run on the teensys and the PDS.

vashmata, 2020-03-14:
I mean more like system-sensitive stuff. IP addresses adn whatnot

NinjaWeeaboo, 2020-03-14:
Ah okay I see, I'll keep this in mind

NinjaWeeaboo, 2020-03-14:
I got seemly all of the ROS nodes and scripts running on the TX2 as systemd services. I was able to successfully run `rosrun ping_acknowledgment ping_response_client.py memes` and get a response from the TX2. With the use of the gui, I was able to see that many rostopics are being subscribed without issue. Only with expected errors in the gui from not having any of the rover's components hooked up to the board. Now I just need to work on setting up the config files for its cameras and other components.

closed 2022-10-31 by VimFreeman