space-concordia-robotics/robotics-prototype #279: Receive 3D LiDAR Data

KoaTheKoala, 2020-01-19:
In order to move between two AR tag posts and avoid obstacles, we need to figure out how to process the data received from the Ouster OS1-16 3D LiDAR.

Link for the OS1 spec sheet (towards the bottom of the page):
https://ouster.com/products/os1-lidar-sensor/

Ouster Studio link:
https://ouster.com/get-started/

- [ ] Setup LiDAR on PC with the Ouster software
- [ ] Figure out the format of the LiDAR data and how to receive the data
- [ ] Receive LiDAR data

Mina-Magdy, 2020-05-26:
The LiDAR has been set up and working properly.
IMU data is being published through the "/os1_cloud_node/imu" topic.
Point Cloud data is being published through the "/os1_cloud_node/points" topic.
For full details, please visit this [link](https://docs.google.com/document/d/13Yz4rR6BTb6egGZ9ver90Zd_LEvbzZgipR2qRectWn4/edit#heading=h.qa0s5ofhd1cg)

KoaTheKoala, 2020-09-19:
Closed to make more relevant and up-to-date issues.

closed 2020-09-19 by KoaTheKoala