space-concordia-robotics/robotics-prototype #280: Build point cloud with 3D Lidar

KoaTheKoala, 2020-01-19:
Before we can start planning a path to move to our goal, we need to build a map with the 3D LiDAR data and identify the rover's location and size. It is possible that Ouster Studio already handles this for us, but it needs to be confirmed.

- [ ] Build a point cloud with the 3D LiDAR data
- [ ] Identify rover location within the point cloud
- [ ] Figure out the size of the rover in the map

vashmata, 2020-01-19:
Why do we need to know the rover's size?

KoaTheKoala, 2020-01-19:
We have be able to estimate the size or some boundary space of the rover so it can properly navigate. If we assume that it's just a dot in the center of the point cloud, it could potentially run into objects or tip over when trying to go around a hill.

vashmata, 2020-01-19:
Ah, ok. Well we will know the rover's actual dimensions because we're the ones making it, that shouldn't be too bad. The point/cloud map should have real-world dimensions and we'll know the rover's size so it should be pretty straightforward.

KoaTheKoala, 2020-09-19:
Closed to make more relevant and up-to-date issues.

closed 2020-09-19 by KoaTheKoala