space-concordia-robotics/robotics-prototype #284: Implement UI testing

adrientremblay, 2020-02-01:
We need to find a way to implement tests for all of the UI features on the base-station.  This will probably be accomplished by creating a bunch of rospy python scripts.

MartensCedric, 2020-02-07:
I would suggest that we expand this to be a PDS mocker. Thoughts @PeterGhimself ?

I don't think it would be a good practice to have a script per UI element. I think we can make a mock per system though

MartensCedric, 2020-02-07:
And this issue would basically be a subtask of that Epic @PeterGhimself

adrientremblay, 2020-02-07:
yeah we definitely need an epic for this

adrientremblay, 2020-02-08:
Publisher name: PDS-Mock publisher

adrientremblay, 2020-03-07:
command to set the voltage mock mode` rosparam set /PDS_mock_mode_voltage <mode>`
where <mode> is the mode (stable, rise, fall) for now

MartensCedric, 2020-07-12:
After my discussion with @ginywiny, here's what we think that the system should do

A high level class/interface should be able to control a float value in three ways.

Either keep it constant, make it rise at a certain rate or make it fall at a certain rate.

In all cases, published values should have noise +- a certain value.

The local_rover.launch files should launch the python node of the pds_local_publisher
The values should all be from rosparameters, written in the local_rover.launch file from <param></param>

The parameters should be the constant value for the constant mode. The noise. The start and rise/fall values for the climbing/falling falling mode.

The system should allow you to control multiple values.
Ex : Voltage Constant, Temperatures increasing, Currents constant.

Voltage should be published to /battery_voltage
Temperatures should be published to /battery_temps (There's 3 temperatures)

closed 2020-10-31 by ginywiny