space-concordia-robotics/robotics-prototype #288: Create Canvas with Rover Bearing

MartensCedric, 2020-02-07:
A canvas to replace the current placeholder should be put in place showing the rover bearing. The direction the rover points at can be displayed by a circle with an arrow. The rover should always be centered at the middle of the canvas.

vashmata, 2020-02-07:
I always get the terms mixed up so here's a link to the definitions of heading and bearing: https://aviation.stackexchange.com/questions/8000/what-are-the-differences-between-bearing-vs-course-vs-direction-vs-heading-vs-tr

MartensCedric, 2020-02-07:
your right @vashmata .

F

well im gonna update it to heading

vashmata, 2020-02-07:
I had some thoughts pop into my head that I'm keeping here for future reference:
1) during autonomy, we will only be setting one GPS goal per leg
2) during extreme delivery, there should be a way to select the desired goal out of all the goals, because this one is what tells the rover which direction to turn to. Ideally it will also be obvious which goal is the target. Speaking of which, the relative bearing between the rover and its goal needs to be displayed somewhere, would this go on the canvas or on the side somewhere?

MartensCedric, 2020-02-08:
[image]
Perhaps we should place the GPS canvas besides this nav screen that @steamypotato did. What do you think @mattsklivas ? @PeterGhimself @vashmata @KoaTheKoala

vashmata, 2020-02-09:
Makes sense, as long as this page is near the camera feeds and the wheel controls so that everyone knows what's going on

PR #338 (2020-05-28): Rover Map
# Assignee Section

## Description
Implementation of a p5.js canvas into the rover page.
The canvas shows the rover position relative to the antenna and its heading.
It is resizable and the view can be dragged.

1. Start the GUI with `rosgui` and `startgui`
2. Go on the rover page
3. Use in terminal 
    ```
    rostopic pub rover_position geometry_msgs/Point "x: 0.0
    y: 0.0
    z: 0.0"
    ```
    This will initialize the rover to that position. You can resend another msg with different coordinates and then the rover will adjust based on the difference.
**THE COORDS ARE IN LONG/L [... cut]
closed 2020-05-28 by AFknUsername