space-concordia-robotics/robotics-prototype #289: Add menu to create and delete goals

MartensCedric, 2020-02-07:
The goals should be added via a menu in the front end. Then the goals should be stored on a ROS node that continuously publishes them at a certain interval (a second ish) for them to be seen by all operators on their GPS map.

- [ ] Goal should consist of name, position and color
- [ ] Goal can be added and deleted via on GUI
- [ ] Goals are published at regular interval and work with multiple operators

vashmata, 2020-02-07:
I've mentioned a few times that parameters could be used. I'm not saying that they should be, but can you explain the reason for using a node that publishes goals as opposed to using parameters?

MartensCedric, 2020-02-07:
@vashmata I'm not sure in what case you would use parameters. The way I think I imagine the system working is the following.

- Operator 1 adds one to multiple goals from GUI which is communicated to node. 
- Node repeatedly publishes all goals to some topic every X seconds. 
- Every operator has their screen updated since their GUI subscribes to the said topic that the node is publishing to.
- Operator 2 can delete a goal as it is no longer used.
- Operators still have their screen updated.

@vashmata I don't see how the parameters could provide the same functionalities.

I'm assuming that it would work in the way that Operator 1 adds a parameter which would then be queried by other operators? If this is the case, I think its simply an inferior use case of the pub/sub model. Further, the parameters do not support custom types while we could make a ros_type "goal" with a name (string), position (x,y) and a color (hex val).

vashmata, 2020-02-07:
I see your point; in my perspective it seemed wasteful to be constantly publishing goal data when it will almost never be needed, whereas with parameters one could query them as needed... But then we'd need to know when a parameter was added or updated so it defeats the purpose. Basically what I realize now is that parameters are best used for initializing things but since the goal feature may need to be updated in real time, pub/sub does make more sense.

That said, you actually can create parameters which are like dictionaries, or even have a parameter which is an array of different types of data. Might be valuable to you for future reference. See [here](http://wiki.ros.org/rosparam#rosparam_set) and [here](http://wiki.ros.org/Parameter%20Server#Parameter_Types)

vashmata, 2020-02-07:
I had some thoughts pop into my head that I'm keeping here for future reference:

1) during autonomy, we will only be setting one GPS goal per leg. Actually, after that the rover will decide on its own how to drive around, maybe we could display its reasoning somehow? Not super critical and more of an autonomy thing
2) during extreme delivery, there should be a way to select the desired goal out of all the goals, because this one is what tells the rover which direction to turn to. Ideally it will also be obvious which goal is the target. Speaking of which, the relative bearing between the rover and its goal needs to be displayed somewhere, would this go on the canvas or on the side somewhere?

MartensCedric, 2020-02-08:
In @steamypotato 's #287 this feature is partially added

MartensCedric, 2020-02-08:
The person taking this issue should continue on his work

AFknUsername, 2020-03-23:
Since fixing #287 involved using a ROS node, I will assign myself this issue.
I have the node working except the colour, but that will be easy to add.
@MartensCedric

closed 2022-10-31 by VimFreeman