space-concordia-robotics/robotics-prototype #291: Show position antenna in GPS canvas

MartensCedric, 2020-02-07:
The antenna should be different from a goal because it has an orientation which should be pointed at the rover. If it is possible to obtain the orientation that the directional antenna is pointing, then we can use that information to display it on the canvas. 

- [ ] Ability to display the antenna on the canvas
- [ ] The orientation of the antenna should update with the canvas

We need to check with @Waleed-HM if this is possible first, if not the issue can be closed as there would be no more value added.

vashmata, 2020-02-07:
It's actually the opposite. Rather than the GUI displaying the direction that the antenna IS facing, the GUI must tell the operators which direction the antenna SHOULD face and it's the operators who adjust the antenna to accommodate that.

vashmata, 2020-02-07:
The ability to display the antenna on the canvas also implies that you need a GUI element that you can use to enter in the coordinates of the antenna. AntennaNode.py expects the user to send it latitude, longitude and initial antenna direction (I believe it's the angle with respect to North but this is where @Waleed-HM can provide information)

MartensCedric, 2020-02-07:
@vashmata so that has nothing to do with this GPS canvas Epic then?

MartensCedric, 2020-02-07:
I repurposed this issue

Waleed-HM, 2020-02-07:
Add me as a participant to this issue.
It's been a while since I worked on this but yeah, that's the gist of it.
The code I wrote before needs the initial antenna direction with respect to north, and using the feedback from the rover GPS chip, it gives you the number you should manually set on the rotator device. Sadly I don't think there is a way to automate this as the rotator device is AC and doesn't have any sort of interface we can use.
Let us discuss this more on Saturday.

MartensCedric, 2020-02-08:
@steamypotato was this done in your #287 ?

vashmata, 2020-03-05:
In #297 I revive the old navigation pane and the antenna director calculations still do stuff.

MartensCedric, 2020-03-05:
Also the new code of @steamypotato once merge will improve the visual
layout of this displayed information

closed 2022-10-31 by VimFreeman