space-concordia-robotics/robotics-prototype #292: Activity Indicator manual control Backend + Firmware

PeterGhimself, 2020-02-07:
1. Backend must be updated to support new request to rover MCU

2. Firmware of rover MCU must be updated to accept requests to turn activity indicator light ON/OFF and change colours

PeterGhimself, 2020-02-07:
I will update the RoverCommandListener to accept the new request, whoever ends up working on the MCU side should make sure to look at my commit first to know what messages to use to recognise the commands, assuming someone else ends up taking up that portion of this task.

PeterGhimself, 2020-02-07:
Starting development on branch [activity-indicator-control-292](/space-concordia-robotics/robotics-prototype/tree/activity-indicator-control-292)

vashmata, 2020-02-07:
Just a note that this is related to #112

closed 2022-10-31 by VimFreeman