space-concordia-robotics/robotics-prototype #297: Investigate issues with custom ROS messages in the GUI

MartensCedric, 2020-02-08:
Investigate and resolve the unreliability of the custom ROS message types in the GUI. 

Make sure you check this : https://github.com/RobotWebTools/roslibjs/issues/254

And once the issue is resolved, these message types should be used throughout the codebase : https://github.com/space-concordia-robotics/robotics-prototype/tree/master/robot/rospackages/src/mcu_control/msg

MartensCedric, 2020-02-25:
So I got it working @vashmata

MartensCedric, 2020-02-25:
@AFknUsername I'm seeing now that Josh did some work

vashmata, 2020-02-25:
A requirement for closing this issue is that it's tested on the Odroid with me present

vashmata, 2020-03-04:
I've tested things out on the odroid, it should be ok. @MartensCedric what would be a smart place to put instructions for running the navigation and antenna node? In our comms setup guides I guess? Should I put it in a readme somewhere?

closed 2022-10-31 by VimFreeman