space-concordia-robotics/robotics-prototype #300: Update git hooks section in README

AFknUsername, 2020-02-13:
The section still uses the old way of committing (using `-m`). 
Should be updated to use whatever @PeterGhimself sent in the slack channel a few weeks back.

- [ ] update readme

PeterGhimself, 2020-02-13:
This is very vague. I just took a look at the README on master and it looks fine. What is the actual specific issue you are trying to define here?

PeterGhimself, 2020-02-13:
Ok, it's not _super_ vague but can you be more specific? Point is I don't see anything wrong with what is currently there

AFknUsername, 2020-02-13:
Essentially you told us to just use `git commit` instead of `git commit -m`, there might've been more but the messages on slack get deleted so I can't see

PeterGhimself, 2020-02-13:
Can you please try to find the exact message because I don't remember this?
If anything I said to do that when doing merge commits, but not in general.

closed 2020-02-13 by AFknUsername