space-concordia-robotics/robotics-prototype #312: Investigate ROS package installation: apt install, git clone, or both?

vashmata, 2020-02-17:
@MartensCedric claims that certain packages, at the very least cv_camera, does not build correctly if it is simply git cloned into rospackages. Furthermore, in the quick setup script which can be used to install ROS and set up a new linux machine for our repo, it calls apt install for certain ROS packages. That said, the ideal scenario is one in which all external packages are linked to our repo and reside in rospackages.

Someone must investigate whether all of our packages can work if we simply git clone them into our repo, or if some packages require installation via apt, or if some packages require both installation methods to be used.

MartensCedric, 2020-05-28:
According to ROS best practices : 

https://www.ros.org/reps/rep-0128.html

git clone > catkin_ws

or sudo apt get install is fine. 

What we are doing is pretty small brain since our rospackages acts as a catkin_ws

With my job, I'm working with multiple ROS project and the catkin_ws approach is more convenient

MartensCedric, 2020-05-28:
Also your cmake targets should go in devel not where we have them, anwyay. That's according to the "recommended layout"

MartensCedric, 2020-05-28:
Removing the good first issue tag because with my suggestion it involves a bit of work to actually make the changes

MartensCedric, 2020-06-13:
Perhaps we can put our dependencies in `package.xml` of task_handler so then we can just run rosdep and have cvcamera and all being downloaded @PeterGhimself

closed 2022-10-31 by VimFreeman