space-concordia-robotics/robotics-prototype #313: Break up the big banks (mcu_control)

vashmata, 2020-02-17:
Currently, the mcu_control package is being used as a large sack of related ROS scripts. While it makes sense for this package to hold the various MCU nodes, it also contains the JoyNode, AntennaNode and other nodes which are indeed directly linked to MCU nodes but perhaps require a new home. Discussions must be had at some point.

MartensCedric, 2020-02-17:
nice meme but let's call it something descriptive

badgerwild, 2020-03-07:
@PeterGhimself @vashmata @adrientremblay @MartensCedric 

 Here is a preliminary package structure for breaking up the mcu_control package into smaller pieces. How does this look:

[image]

adrientremblay, 2020-03-07:
On my branch I'm working on a PDSMockPublisher.py which will create nodes to mock publish voltage and temperature.  Where should that eventually go?

badgerwild, 2020-03-07:
Here is the link to the doc to edit the diagram: 
https://drive.google.com/a/spaceconcordia.ca/file/d/1WcYjL5RZ45hbZvbL9iNlJp2p-XnJ-Lqb/view?usp=sharing

vashmata, 2020-03-08:
> On my branch I'm working on a PDSMockPublisher.py which will create nodes to mock publish voltage and temperature. Where should that eventually go?

So the issue with your mock publisher is that it will probably be using custom messages defined in mcu_control. One issue @badgerwild needs to tackle is to figure out how to reference a message type defined in a different package, but now that I think about it this should be trivial since we do this all the time when we use std_msgs and whatnot.
@adrientremblay up to you in the end, either you put it in mcu_control or you create a new package for mocking MCU data. If you go with the latter, you'll need to deal with the problem mentioned above, unless @badgerwild figured it out first.

MartensCedric, 2020-03-14:
@vashmata I was thinking more like each PDS, Arm, Navigation and Science to have their own package. IK would go in Arm. Pds stuff with PDS. RoverNode + Nav stuff in Navigation and science in science.

vashmata, 2020-03-14:
I was thinking more in terms of layers of abstraction; the mcu nodes all behave similarly and all interact directly with mcus so I would keep them together. This also makes more sense since they would all be using the serial module. Whereas something like the ps4 controller is high level so it should be in its own bubble

MartensCedric, 2020-03-21:
I understand the point and I think it is decent. What I propose is organize the packages as modules. Our rover is modular. If we don't have science running it makes sense to not using the science package. In my description of the organization packages are modules that we add to the rover. What do you think is more appropriate? You have the final words vashmata @PeterGhimself

badgerwild, 2020-04-04:
@vashmata @MartensCedric  @PeterGhimself 
Following Cedric's comment here is an alternative package organization scheme. What do you think, does this make sense, or is the previous structure better? 

I also noticed that a lot of the files are named using camel case. I think they should be renamed to snake case for consistency. Is that best done here, or would it be better to make that part of a re-factoring issue, since they are likely used in other places on the rover code? 
 
[image]

Link for editing : https://drive.google.com/open?id=1Ksf8jPU301tvu0Zw8nZEueJCNwx1TDKQ

MartensCedric, 2020-04-18:
@badgerwild  I like this package structure a lot. There's some small things like mcu_serial should not be in PDS. It's a shared file IIRC with multiple things (perhaps a robotics_util?) @vashmata What do you think josh?

vashmata, 2020-04-26:
Yeah as Cedric mentioned mcu_serial should go into a utilities box or something. As far as I'm aware it doesn't do anything ROS related so maybe we can put it with the rest of the python utility submodules? @PeterGhimself pls advise

Also I'm fine with RoverNode, JoyNode and Joy_class being inside Navigation but maybe we should consider having a separate package for them. Opinions?

Finally, for naming conventions see [this link](http://wiki.ros.org/ROS/Patterns/Conventions#Naming_ROS_Resources) from the ROS wiki. Perhaps it would be best to make a new issue for renaming the files.

MartensCedric, 2020-06-11:
> Yeah as Cedric mentioned mcu_serial should go into a utilities box or something. As far as I'm aware it doesn't do anything ROS related so maybe we can put it with the rest of the python utility submodules? @PeterGhimself pls advise
> 
> Also I'm fine with RoverNode, JoyNode and Joy_class being inside Navigation but maybe we should consider having a separate package for them because drive controls and navigation can be considered 2 different things. Opinions?
> 
> Finally, for naming conventions see [this link](http://wiki.ros.org/ROS/Patterns/Conventions#Naming_ROS_Resources) from the ROS wiki. Perhaps it would be best to make a new issue for renaming the files.

I still think navigation can be seen as one thing. Eventually the drive controls are supposed to abstract manual and autonomous control in one interface.

closed 2022-10-31 by VimFreeman