space-concordia-robotics/robotics-prototype #324: Increase the robustness of the mcu_control scripts

MartensCedric, 2020-02-25:
These scripts often crash and the node should be able to survive serial issues

vashmata, 2020-02-25:
I think it makes sense for the node to close down if serial issues arise. However it should be verbose and publish a message that can be caught by the GUI when it happens

MartensCedric, 2020-02-29:
I disagree, serial issues are common and it makes no sense to shutdown at the moment that one command was corrupted. The node should be robust enough to be able to survive badly formatted things such that its ready to accept the next valid command. It should definitely log bad

MartensCedric, 2020-02-29:
Upon further investigation it appears that most nodes already try catch most cases but theres 1-2 places where the person forgot

MartensCedric, 2020-02-29:
Looking at this log

```
Just pasting a different crash : Maybe related, maybe not : INFO] [1582665360.080515]: got raw data: ASTRO Linear Velocity: -0.00 m/s RotaASTRO GOT: 26:0

[WARN] [1582665360.138258]: GPS data unavailable
[INFO] [1582665360.143210]: received: 27:0 command from GUI, sending to rover Teensy
[INFO] [1582665360.156355]: got raw data: Desired Velocities: 135.31, 135.31, 135.31, 135.31, 135.31, 135.31

[INFO] [1582665360.215620]: received: 28:0 command from GUI, sending to rover Teensy
[INFO] [1582665360.286379]: received: 29:0 command from GUI, sending to rover Teensy
[WARN] [1582665360.344597]: GPS data unavailable
[INFO] [1582665360.353643]: received: 30:0 command from GUI, sending to rover Teensy
[INFO] [1582665360.360088]: got raw data: Desired Velocities: 150.92, 150.92, 150.92, 150.92, 150.92, 150.92

[INFO] [1582665360.422127]: received: 31:0 command from GUI, sending to rover Teensy
```
**[INFO] [1582665360.429715]: got raw data: ASTRO LinASTRO GOT: 31:0**
```

[INFO] [1582665360.493040]: received: 32:0 command from GUI, sending to rover Teensy
[INFO] [1582665360.562823]: received: 33:0 command from GUI, sending to rover Teensy
Traceback (most recent call last):
File "/home/odroid/Programming/robotics-prototype/robot/rospackages/src/mcu_control/scripts/RoverNode.py", line 322, in
publish_nav_states(feedback)
File "/home/odroid/Programming/robotics-prototype/robot/rospackages/src/mcu_control/scripts/RoverNode.py", line 204, in publish_nav_states
heading,gps = message.split(' -- ')
ValueError: too many values to unpack (expected 2)
(venv) odroid@odroid:~$
```

Most likely some packet `ASTRO LinASTRO GOT: 31:0` got corrupted so the whole thing crashed. Expected behavior should be to just ignore this packet and move on

vashmata, 2020-02-29:
Yeah maybe it was missed but the expected behaviour is that if a message contains "ASTRO" twice, it should ignore it as a corrupted message.

closed 2022-10-31 by VimFreeman