space-concordia-robotics/robotics-prototype #342: Emphasize the warning when voltage or temperature become in a warned range

MartensCedric, 2020-03-14:
Currently, in the GUI, when the battery voltage or the PDS temperatures are in a dangerous range, the operator gets a model popup. However, when the values are close to the danger value, it is in a warning threshold. Other than changing the text to yellow, not much is done. This issue would require the assignee to make it more clear to the operator that it's in the warning threshold. This could be done by changing the background of some component in the navbar. Alternative solutions are also welcome.

AFknUsername, 2020-03-22:
Alternative solution: Make a huge rainbow arrow fade in and out pointing at the temp value

MartensCedric, 2020-06-13:
We found a library called notify.js

closed 2022-10-31 by VimFreeman