space-concordia-robotics/robotics-prototype #359: Error in setup script

AFknUsername, 2020-05-06:
The automatic setup script adds two lines to `.bashrc` which cause errors with ROS. The script needs to be updated to not include those lines. 

- [ ] Modify script to source the lines without adding them to `.bashrc`

AFknUsername, 2020-05-06:
The easiest fix I think is to keep a local copy of the install script for ROS and modify it to not include the lines, and maybe edit other things too if necessary.

MartensCedric, 2020-05-06:
> The easiest fix I think is to keep a local copy of the install script for ROS and modify it to not include the lines, and maybe edit other things too if necessary.

Seems like the correct solution to me.

AFknUsername, 2020-05-14:
@MartensCedric this needs to be tested, but I'm pretty sure it should work.

PR #366 (2020-05-30): Fix environment setup 359
# Assignee Section

## Description
Fixes a serious ROS/Catkin configuration error caused by the third-party ROS install script `install_ros_kinetic.sh`. 
The fix involves removing the erroneous addition of some commands to the `.bash_rc` and also removing the catkin workspace setup since we have our own process. 

- Removed bad sourcing
- Fixed catkin workspace setup

### Steps for Testing

1. Either setup a fresh VM or create a new user account in Ubuntu
2. Run the setup script: `./EnvironmentSetup.sh`
3. Check if the GUI works: `rosgui` in one terminal followed by ` [... cut]
closed 2020-05-30 by AFknUsername