space-concordia-robotics/robotics-prototype #360: Fresh clone attempts to get joint_state_publisher but its not even in master

MartensCedric, 2020-05-06:
Joint state publisher was added to `.gitmodules` in #278, yet when you try to clone the project you will get 

```
Checking connectivity... done.
Submodule 'cv_camera' (https://github.com/OTL/cv_camera) registered for path 'robot/rospackages/src/cv_camera'
fatal: no submodule mapping found in .gitmodules for path 'robot/rospackages/src/joint_state_publisher'

```

Despite it not being in master

PeterGhimself, 2020-05-18:
I can't reproduce this, can anyone else?

```
(venv) beep@boop:~/test$ git clone https://github.com/space-concordia-robotics/robotics-prototype
Cloning into 'robotics-prototype'...
remote: Enumerating objects: 189, done.
remote: Counting objects: 100% (189/189), done.
remote: Compressing objects: 100% (144/144), done.
remote: Total 24609 (delta 84), reused 102 (delta 40), pack-reused 24420
Receiving objects: 100% (24609/24609), 179.75 MiB | 676.00 KiB/s, done.
Resolving deltas: 100% (16184/16184), done.
Checking connectivity... done.
```

PeterGhimself, 2020-05-18:
With the recursive option, I can reproduce it after all.

```
(venv) beep@boop:~/test$ git clone --recursive https://github.com/space-concordia-robotics/robotics-prototype
Cloning into 'robotics-prototype'...
remote: Enumerating objects: 189, done.
remote: Counting objects: 100% (189/189), done.
remote: Compressing objects: 100% (144/144), done.
remote: Total 24609 (delta 84), reused 102 (delta 40), pack-reused 24420
Receiving objects: 100% (24609/24609), 179.75 MiB | 220.00 KiB/s, done.
Resolving deltas: 100% (16184/16184), done.
Checking connectivity... done.
Submodule 'cv_camera' (https://github.com/OTL/cv_camera) registered for path 'robot/rospackages/src/cv_camera'
fatal: no submodule mapping found in .gitmodules for path 'robot/rospackages/src/joint_state_publisher'
```

PR #365 (2020-05-18): [#360] Remove modules which were supposed to be added in another branch instead of master
# Assignee Section

## Description
Removed folders that were accidentally added in another PR (They shouldnt have been added)
This makes the clone fail as the folders were added (Accidentally) without updating the .gitmodules file.

### Steps for Testing
1) There's not much you can do to test because it needs to be in master to fix the clone lmao

closes #360 

The approval from all software team leads is necessary before merging.

# Reviewer Section

Aside from local testing and the General Integration T [... cut]
closed 2020-05-18 by MartensCedric