space-concordia-robotics/robotics-prototype #362: Teensyduino and libraries instructions not documented

MartensCedric, 2020-05-09:
We should have a section in the README explaining how to setup teensyduino as well as which libraries must be downloaded. Note that we should think how we would automate this.

PeterGhimself, 2020-05-14:
Isn't that explained [here](https://github.com/space-concordia-robotics/robotics-prototype/tree/master/robot/rover#teensy)?

MartensCedric, 2020-05-14:
> Isn't that explained [here](https://github.com/space-concordia-robotics/robotics-prototype/tree/master/robot/rover#teensy)?

Not quite,

When you first run the project it won't compile. You will be missing libraries. To get them, you have to search your error, find what libraries is missing, download it, rinse and repeat 10 times. The list of used libraries could be documented and a link to the document you points should be in the wiki or README if we want people to see it.

PeterGhimself, 2020-05-14:
Maybe we should just have like a `requirements.txt` sort of file to contain every dependency line by line and their version. Perhaps that could allow for some easier custom automation, if there is no existing framework to help automate. Btw, the libraries your refering to need to be downloaded through arduino IDE?

MartensCedric, 2020-05-14:
>  Btw, the libraries your refering to need to be downloaded through arduino IDE?

Yup, since its compiled with Arduino afaik, but @steamypotato had some scripty ideas

vashmata, 2020-05-14:
Arduino libraries can be downloaded from github and dropped into the libraries folder. That said, I haven't tested this all that much in linux, we will need someone to uninstall the libraries and try it out. It's possible that not all our libraries will "like" this approach.

vashmata, 2020-05-14:
As it is I got the impression that some of the libraries cedric is referring to don't play nicely, i remember my laptop being the only one capable of downloading code to the rover teensy at a few points

PeterGhimself, 2020-05-14:
> i remember my laptop being the only one capable of downloading code to the rover teensy at a few points

Yeah at URC 2019 that was the case I remember, but then I fixed the issue for my laptop. It's been a while so I'd have to dig a bit to see how I fixed it.

MartensCedric, 2020-05-24:
These are the libraries I have installed, the project probably needs less than that. The assignee of this task should determine which ones are actually needed.

[image]

[image]
[image]

steamypotato, 2020-05-24:
Minimum to get the 'important' code in Rover working (not counting test code and dev code as these are subject to a refactoring issue)
- Adafruit Motor Shield V2 Library -> 1.0.9
- ArduinoBlue -> 3.0.1
- LSM303 -> 3.0.1
- SparkFun I2C GPS Reading and Control -> 1.0.2 (newest version 1.0.3 doesn't compile with teensy compiler)
These are the external ones you gotta manually download

MartensCedric, 2020-05-26:
So on my images I was missing Adafruit Motor Shield V2 Library -> 1.0.9

As @steamypotato correctly pointed out

PR #392 (2020-06-16): Teensyduino libs install 362
# Assignee Section

## Description
The CMake script in `robot/rover` installs all needed Teensyduino libraries, using `install_libraries.sh` in `robot/rover`. By default, Arduino libraries are installed in `~/Arduino/libraries`, so this script won't work if someone changed the default installation directories of the Arduino IDE.

### Steps for Testing
1. Delete all libraries in `~/Arduino/libraries` (except for unrelated libraries installed for personal use)
2. Go to `robotics-prototype/robot/rover` 
3. `mkdir build && cd build`
4. `cmake ..` and ensure that the li [... cut]
closed 2020-06-16 by steamypotato