space-concordia-robotics/robotics-prototype #367: Redesign the way changeListenerStates function in JavaScript

MartensCedric, 2020-05-16:
Currently, the listener state system checks on the specific page that the listener is on to do some functions. This creates dirty code and undesired behaviors on pages that won't have the listener (navigation, multiple camera streams). 

A suggestion could be to make the listener system into one template, but other solutions are welcome.

vashmata, 2020-05-31:
If I understand correctly, the issue is that if you're on, say, the arm page but want to send a command to the wheels, the js may not function, because topics and such only get initialized on the wheels page?And therefore you're proposing a template that initializes all of the topics and services and whatnot that gets loaded on every page?

MartensCedric, 2020-05-31:
No, this has nothing to do with ros. There's the rover listener thing and
arm listener thing (in navbar.js) i think, and they have if statements that
depend on the page, but they assume every page has these so it fucks for
stream and nav page since they dont have arm or rover listener. something
like that

closed 2022-10-31 by VimFreeman