space-concordia-robotics/robotics-prototype #368: Trouble reading from serial port

PeterGhimself, 2020-05-19:
While testing PRs using the simulated GIT solution, I encountered the message `trouble reading from serial port` when trying to test movement commands from GUI to rover.

Steps to reproduce:

1. checkout master, pull and make sure everything is up to date
2. rosgui
3. startgui
4. in util/rosRoverStart: roslaunch local_rover.roslaunch
5. go to rover page, activate cmd listener, activate rover
6. use wasd keys to simulate movement
7. observe terminal for output from python command listener process

PeterGhimself, 2020-05-19:
@vashmata any idea what might be causing it? I believe you are the last one to touch those lines (`git blame` ftw)

vashmata, 2020-05-31:
You'd have to read the listener node's code to see what triggers this error. If my memory is correct (often it isn't), this error happens when the listener script detects that the message coming in from the serial buffer is garbled (for example, it says ARM twice or it doesn't say ARM at all, the first case meaning that 2 messages were smashed into one).

- One cause may be the fact that I removed the serial.flush() stuff.
- Another cause may be due to the rate at which messages are being sent and received.
- I've also noticed that if you have the arm and the wheels teensies both connected to the rover and start the scripts in a particular order, it has a higher rate of failure.

closed 2022-10-31 by VimFreeman