space-concordia-robotics/robotics-prototype #369: Javascript coding standards enforcement

PeterGhimself, 2020-05-22:
- [ ] Choose appropriate solution
- [ ] Implement solution

PeterGhimself, 2020-05-22:
I'm going to suggest using [this](https://github.com/standard/standard#is-there-a-git-pre-commit-hook) 
 pre-commit hook as a solution, that way we don't have to worry about setup configuration for our varied editor/IDEs

AFknUsername, 2020-06-15:
@PeterGhimself seems like standard uses Node.js and npm.
To implement it we would need to install both packages into our repo, adding clutter. 
I'm ok with implementing it, but I would like your input first.

MartensCedric, 2020-06-15:
We both agreed a while back that we would really try to avoid using Node
and npm since we both had a poor experience with it previously on unrelated
projects. If you could find an alternative, that does not depend on
node/npm then it would be great

AFknUsername, 2020-06-15:
@MartensCedric
I guess that rules out standard then, they only have IDE specific packages, or node.
Also, so far I've noticed that a linter might be problematic on the JS files as we use a lot of separate files and the report ends up being absolutely swamped with 'var name' not defined messages because it doesn't know that the files are linked through html.

MartensCedric, 2020-06-15:
Fyi Webstorm has a pretty good built in linter, I used it in update nav to
remove useless stuff added in the pr

AFknUsername, 2020-06-15:
The built in linter is actually standard linter

AFknUsername, 2020-06-15:
So from the research I'm doing, it would be hard to integrate a JS linter into our codebase since we have so many interdependent files with many implicitly imported variable through html. 
Although a "per-file" formatter is possible.

closed 2022-10-31 by VimFreeman