space-concordia-robotics/robotics-prototype #374: Refactor the PDS Node

MartensCedric, 2020-05-27:
The PDS node has extremely odd for loops and could definitely be improved through some refactoring. This task is to rewrite the structure of the PDS node so that it has high quality code.

vashmata, 2020-05-31:
I don't remember if it's the case with the PDS node but a bunch of the MCU nodes are a bit weird simply due to me trying to figure out how to get them to behave as publishers, subscribers, services and whatnot all at the same time. So if you happen to start working on very big structural changes try to be thorough about testing everything.

also FYI #207 is something you might be interested in tackling at some point as it's related to refactoring these nodes :)

AFknUsername, 2020-05-31:
Thanks for the heads up @vashmata 
Looks like #207 is already being worked on, but I'll definitely check if there is anything else to refactor.
Also, since we got the custom message types figured out, it's a really good opportunity to rework the code

vashmata, 2020-05-31:
yeah my bad I just noticed this https://github.com/space-concordia-robotics/robotics-prototype/pull/282
I've just been going through all my emails that i accumulated in the past 2 weeks haha

PR #383 (2020-06-10): [#374] Refactor PDS Node
# Assignee Section

## Description
Refactoring of the PdsNode.py

closes #374 

The approval from all software team leads is necessary before merging.

## Testing

1. Take an Arduino and upload the PDS code (Don't forget to switch from teensy device to Arduino lmao)
2. `cd robot/rospackages`, then build new messages with `catkin_make`. If you get some errors from `catkin_make` it may be because of missing requirements, if so try `rosdep install --from-paths src/ --ignore-src`.
3. `roscore`
4. In a new terminal: `rosrun mcu_control PdsNode.py usb`
5. Check that you rece [... cut]
closed 2020-06-10 by AFknUsername