space-concordia-robotics/robotics-prototype #377: Remove useless C++ files

MartensCedric, 2020-05-30:
We have files like "ideas.h" and "notes.h". There's folders with example code which we will clearly never use. Looking through this document might be useful : https://docs.google.com/document/d/1MZaEAh4fQXDNMqdmfaYagB0u23k_reNE0J9pfIDxf2A/edit#

The assignee must remove as much useless files, projects and example projects as he can in the C++ code.

MartensCedric, 2020-05-30:
@steamypotato for you to :eye: :lips: :eye:

vashmata, 2020-05-31:
wow stop calling me out

vashmata, 2020-05-31:
Your issue description makes no mention of the fact that whoever takes this issue upon themselves had better check before deleting anything 🔪 🔪 🔪

MartensCedric, 2020-05-31:
Fauzi said we can remove demo projects (source the SQA doc and telegram chat with him)

vashmata, 2020-05-31:
Yeah I'm just making sure it's clear when it comes time to delete my stuff 👀

PR #403 (2020-06-16): Removed Useless C++ Files
# Assignee Section

## Description
I removed a bunch of useless C++ files such as `ideas.h` or `notes.h` as well as a bunch of test projects. This makes the repo much cleaner.

### Steps for Testing
Ensure that you have the arduino IDE setup from the wiki and can build with cmake.
1. In rover/, do `mkdir build`
2. `cd build`
3. `cmake ..`
4. `make`

Everything still builds because those files were indeed, useless

closes #377

The approval from all software team leads is necessary before merging.

# Reviewer Section

Aside from local testing and the General Integratio [... cut]
closed 2020-06-16 by MartensCedric