space-concordia-robotics/robotics-prototype #378: Get rid of the includes in the middle of the c++ files

MartensCedric, 2020-05-30:
In a few instances, there is some C++ includes at the middle of the file. This is not good practice and should be ended. 

More information on which files are problematic can be found here. 
https://docs.google.com/document/d/1MZaEAh4fQXDNMqdmfaYagB0u23k_reNE0J9pfIDxf2A/edit#

The assignee should make sure that all of them are fixed

vashmata, 2020-05-31:
In some locations this may be due to (bad) code design that requires the includes to be placed after certain other lines of code in order for things to function correctly.

MartensCedric, 2020-06-04:
> In some locations this may be due to (bad) code design that requires the includes to be placed after certain other lines of code in order for things to function correctly.

Yep, and I can almost bet that this is because of #379, which might need to be done first oof

closed 2022-10-31 by VimFreeman