space-concordia-robotics/robotics-prototype #381: Camera template doesn't always display properly on first opening

MartensCedric, 2020-05-30:
Steps to reproduce the bug

1. rosgui
2. startgui
3. roslaunch local_rover.launch from (rover/util/rosRoverStart)
4. Select camera
5. Turn on Camera (Nothing changes)
6. Refresh
7. Select Camera (works)

Very interestingly,
8. Open new tab
9. Select camera (works)

It seems that this issue happens only the first time the camera is selected. I don't know if it is a GUI or python issue.

MartensCedric, 2020-05-30:
might be a good issue for you @Gmartinica , you miight have to look into task handler code though but you are familiar a bit with the camera

Gmartinica, 2020-06-06:
Oh I see, so if you have the GUI already opened and you follow the steps it does not work, I was trying opening the GUI after doing the first 3 steps.

MartensCedric, 2020-06-06:
Could you say exactly what steps reproduced the bug? Maybe I didnt put them correctly but just in case, it would be nice to have the two sets of steps

Gmartinica, 2020-06-06:
Just to make sure, after running: 
1. rosgui
2. startgui

Do you reload or open a new tab for the GUI to be visible? Because the reason it was not working in my computer is because I was trying running:

1. rosgui
2. startgui
3. Opening the GUI in a new tab
4. Killing both processes
5. Follow your instructions 

But this closes the websocket server and never opens it again, you have to reload the page to establish a connection again.

PeterGhimself, 2020-06-13:
Just tried @MartensCedric original steps, cannot reproduce this bug

Gmartinica, 2020-06-13:
Update on the error. I thought I had reproduced the error but actually I was closing the server connection and was not updating the GUI, so of course that did not work. In short, following @MartensCedric steps I could not reproduce the bug either

MartensCedric, 2020-07-09:
So apparently this is not reproduceable so ill close but Ill reopen if it happens again

MartensCedric, 2021-01-06:
We got this issue on the Odroid again @MewtR and @KoaTheKoala. We don't know what are the steps yet, I'm reopening this issue and hopefully we can find the steps

closed 2022-10-31 by VimFreeman