space-concordia-robotics/robotics-prototype #382: Add upload to teensy option in CMAKE or make a script for it

MartensCedric, 2020-05-30:
The CMake script builds the C++. It should have an option to upload it to a teensy, or we should have a script somewhere that does just that. I think @PeterGhimself had some work from SOEN422, maybe he can comment it here.

steamypotato, 2020-05-30:
@MartensCedric the end result would probably be `make install`, where this command takes binaries (the .elfs which will get converted into .hex) and configures them/sends them to a desired location, such as a Teensy Board. I'd be pretty comfortable doing this, but I don't have a Teensy board lolz

MartensCedric, 2020-05-30:
@steamypotato you can get one but its the cheapest thing. 30$ USD + you need a 5$ cable with shipping it comes to like 50-60 CAD. We have a bunch spread across mtl though that maybe you can meet up with someone who borrowed one from the club. See this https://docs.google.com/spreadsheets/d/1JO1WpAUDEGv29qkNKcEJh-EtENpHMGB2UkPywm21ep0/edit

MartensCedric, 2020-05-30:
it has to be a teensy 3.6

PeterGhimself, 2020-05-30:
@Waleed-HM has an extra one he doesn't need, I think Valentina is going to pick it up soon. Perhaps we could forward it over to @steamypotato ?

steamypotato, 2020-05-30:
@MartensCedric also, shipping from US during pandemic + not amazon prime = at least 1 month shipping time

MartensCedric, 2020-05-30:
CAD for a teensy3.6 + cable + shipping

MartensCedric, 2020-05-30:
> @MartensCedric also, shipping from US during pandemic + not amazon prime = at least 1 month shipping time

I bought one like 2 weeks ago and it took like 4 days (bought from robotshop), although it cost me like 56$ cad ish

MartensCedric, 2020-05-30:
I would recommend you borrow one, if you live close to someone in the team

PeterGhimself, 2020-05-30:
Yeah I wrote some code way back which did this for atmega328p but I did not write it to be modular and it definitely could stand to be improved in multiple ways, though it might be useful for this issue.

As it was written, it was meant only to be run on .cpp files, and only one at a time (I think).
```
#!/usr/bin/env bash

if [ "$#" -lt 2 ]; then
    echo "Illegal number of parameters"
    exit 1
fi

c_file=$1
arduino_port=$2

avr-gcc -Os -Wall -mmcu=atmega328p $c_file.cpp -o $c_file.o &&
avr-objcopy -O ihex -j .text -j .data $c_file.o  $c_file.hex &&
/home/peter/Downloads/arduino-1.8.7/hardware/tools/avr/bin/avrdude -C /home/peter/Downloads/arduino-1.8.7/hardware/tools/avr/etc/avrdude.conf -v -p atmega328p -c arduino -P $arduino_port -b 57600 -D -Uflash:w:$c_file.hex:i
```

PeterGhimself, 2020-05-30:
Essentially it compiles the code, converts it to hex, and then uploads to the port passed as the second arg `arduino_port`, that's why I called it `chu.sh` 😎

PR #419 (2020-10-03): Cmake upload teensy 382
# Assignee Section

## Description
Using CMake, upload the compiled C++ to a Teensy board, using the Teensy Loader CLI program from https://github.com/PaulStoffregen/teensy_loader_cli . It uploads one project at a time, although since the Rover uses 3 Teensys, this can easily be expanded to upload everything at once. Requires a Teensy board to test.

### Steps for Testing
1. Ensure that `rover/config/.bash_aliases` is being sourced, preferably from `~/.bashrc`
2. Go to `robot/rover`, and run `./install_teensy_loader_cli.sh`
3. In `rover`, run `mkdir build && cd build`
4 [... cut]
closed 2020-10-03 by MartensCedric