space-concordia-robotics/robotics-prototype #387: Twisted Security Alerts

MartensCedric, 2020-06-04:
I added this library to fix the rosbridge venv issue, but it ended up being useless. Now we have security alerts for this library.

I'm 95% sure this library can be removed safely. If this is not the case then we should just upgrade it so we don't get these security alerts on our repo and everytime we push.

PeterGhimself, 2020-06-05:
I looked into this a while ago, I tried updating it and this resulted in some dependency clashes making it not possible to run the GUI.

AFknUsername, 2020-06-13:
@MartensCedric I tried to uninstall Twisted using pip and it broke ROS. It uses it for some things like logging. 
@PeterGhimself I upgraded using `pip install -U Twisted` and it worked fine. GUI seems to be fine as well.

Thoughts?

MartensCedric, 2020-06-13:
The library should not even be used. Idk why it broke ROS for you but this is something I added in September. 

https://github.com/space-concordia-robotics/robotics-prototype/pull/154

And this PR actually did not fix the issue so it should be safe to just undo these changes.

AFknUsername, 2020-06-13:
This is what happens when I install latest Twisted using pip3: 
[image]

Side note: we prob can remove the attrs requirement from requirements.txt since Twisted installs it.

PeterGhimself, 2020-06-14:
closing via #398

closed 2020-06-14 by PeterGhimself