space-concordia-robotics/robotics-prototype #416: Find formatters and linters for JS and C++

AFknUsername, 2020-06-15:
We need them to enforce code formatting and to check code quality.

AFknUsername, 2020-06-16:
@MartensCedric @PeterGhimself 
Here is what I have found: 
JS: jsbeautifier (it just formats the code)
C++: clang-format for formatting (we can choose a standard) and cpplint for linting (it uses the google standard)
And just for completeness:
Py: yapf and pylint

All of them can be run from command line and the linters have helpful outputs which can be used to increase code quality.

MartensCedric, 2020-06-18:
Just to point out that there's a tool called Clang-Tidy which is the built-in CLion linter

MartensCedric, 2020-06-18:
https://clang.llvm.org/extra/clang-tidy/

PeterGhimself, 2020-06-18:
Having the CLI option first is great, but I think we should add support for a text editor for this issue. I think sublime3 is a good option since it's free, lightweight, configurable, and it is actually sensitive to external changes (like when switching between branches, the code actually updates itself, as opposed to in atom where it does not).

As for standards, I think we'd have to look into the c++ ones more, making sure that it's appropriate in an embedded context (and preferably adhering to MISRAC guidelines).

As for python, I believe the pylint file we have conforms to PEP8 which is the most standard convention today.

MartensCedric, 2020-06-18:
I don't know if it matters, but we compile using GCC, now I don't know if performing to a Clang style guide can break the code, since its meant to be on a different compiler :shrug:

MartensCedric, 2020-06-18:
> I don't know if it matters, but we compile using GCC, now I don't know if conforming to a Clang style guide can break the code, since its meant to be on a different compiler

this is kind of why we shouldnt put it on a hook immediately

AFknUsername, 2020-06-18:
@MartensCedric I tried clang-tidy and it did not work because it needs an entire project to be open or something.
@PeterGhimself I really don't think sublime is the way to go. If you use the atom git package (comes preinstalled) it's really easy to update between branches and stuff. For python we're already using PEP8 (I remove the 80 character limit on all formatters because it's stupid and useless and looks bad when everything is folded. (Also as Linus himself said, it's useless)

AFknUsername, 2020-06-30:
> I don't know if it matters, but we compile using GCC, now I don't know if conforming to a Clang style guide can break the code, since its meant to be on a different compiler 🤷

Seems like clang defaults to GCC's library when being used on linux.

MartensCedric, 2020-07-04:
Sounds good to me 
     :sunglasses:
:+1:

closed 2022-10-31 by VimFreeman