space-concordia-robotics/robotics-prototype #417: Create pre-commit hook to run formatting and linter checks on committed files

AFknUsername, 2020-06-15:
To enforce code formats, we can implement formatters that run on each file pre-commit. 
To enforce code quality standards, linters will be used to assess the code and fail if the code is not good enough.

PeterGhimself, 2020-06-16:
Rather than use pre-commit hooks I'm thinking perhaps providing an adequate solution in IDE solution would be the better approach. The prospect of coding and only having checks done when I'm ready to commit seems less appealing then live squiggly lines as I go.

For the formatter I would also think it would be nicer to link it to a shortcut in IDE to apply to the file you're opening.

We had this pretty much how I described setup before for atom, before we lost all hope for using atom moving forward. On save (ctrl + s) it would run the auto-formatter yapf, and standard-js for js. Linter would just always be active.

AFknUsername, 2020-06-16:
I guess we could do both. I think it would still be more appropriate to enforce it from an "external" perspective. We can't rely on everyone setting up their IDEs and actually formatting their code.

PeterGhimself, 2020-06-18:
I don't mind having it maybe evenutally added in hooks, but I think it would be best to put that on hold for at least a few months after #416 is resolved. It would be safer to make sure whatever new formatters/linters we enforce don't cause any unforeseen problems that then introduce automated nuisances in our workflow.

For instance: imagine you write some code, the formatter re-formats it and somehow that actually ends up breaking the code. We didn't catch it for the PR for #416 because certain edge cases weren't encountered during testing.

AFknUsername, 2020-06-18:
@PeterGhimself That's why this issue is blocked by the previous one.

PeterGhimself, 2020-06-19:
Right, and that's good that you showcased the dependency there, but I am saying we shouldn't rush this issue even after #416 is closed. Might be appropriate to agree to a certain time period (like 2 months for example) after #416 is closed before moving forward with this one. That make sense?

AFknUsername, 2020-06-19:
@PeterGhimself yeah that makes sense

closed 2022-10-31 by VimFreeman