space-concordia-robotics/robotics-prototype #418: Create a docker image of the environment for convenience (and potentially for travis)

AFknUsername, 2020-06-18:
Creating a docker image of our environment would make setting up the env a painless process (just launch the image). This would allow for more rigorous testing and bold modifications because people wouldn't be afraid to nuke their install 🙂 

Also potentially travis would be able to launch the docker image, which would greatly reduce the time per build since currently about 2.5 mins is dedicated to installing apt packages.

MartensCedric, 2020-06-18:
@adrientremblay please share what you did so far

adrientremblay, 2020-06-18:
https://github.com/adrientremblay/dockerfiles/tree/master/ros-sc

The dockerfile is very messy but should work.

AFknUsername, 2020-06-18:
@adrientremblay wanna commit it to a new branch? I'll clean it up and play around with it (maybe to set a custom user based on a prompt or os $user env).

adrientremblay, 2020-06-18:
@AFknUsername Allright I created a branch. https://github.com/space-concordia-robotics/robotics-prototype/tree/docker-418.  I also made a 'docker' branch because I forgot we had to add issue #s in the name.

AFknUsername, 2020-06-19:
That moment when typing $USER in the commit message replaces it with the value... 💩

MartensCedric, 2020-06-19:
10 IQ

MartensCedric, 2020-06-19:
If your repo is setup correctly, it should block you from making an illegal
branch name. Make sure your hook is setup

AFknUsername, 2020-09-19:
@MewtR 
Something I was planning on doing is adding the possibility to pass the $USER argument to the docker build so that the people using it have a user with the same name inside docker.

MewtR, 2020-09-19:
I'll look into it but from what I've read we might have to force people to use their own user inside the containers. This issue has more to do the the GUI though, see [this](https://wiki.ros.org/docker/Tutorials/GUI) if you're interested.

MewtR, 2020-09-20:
Far from perfect but I think I have a basic setup working. I see some messages in the log console but I'm not sure if they're related  to a lack of hardware or if I messed something up. Anyone down to help me test this?

AFknUsername, 2020-09-20:
I'm willing to help you out, but you will have to write down some testing steps and expected behaviour. 
Or if you think you just need to finalize some things and want feedback, feel free to open a PR and request some feedback.

MewtR, 2020-09-20:
I wrote some instructions on how to run it in the REAME.md in the docker directory. Basically what I want to know is:
* Does it actually work for anyone else besides me
* Does the project run as expected. I'm not familiar with the project enough to be able to answer this question.

AFknUsername, 2020-09-20:
I merged master into the branch because it was stale.

AFknUsername, 2020-09-20:
I looked into it and found a couple of things.
Less important:
1. docker-compose needs to be run as sudo. Most people don't know that they can add themselves to the docker group
2. Maybe put detailed instructions on how to install docker and docker-compose. Again, people are dumb and might just do `sudo apt install docker`

More important:
No the websocket error is not normal. You will need to figure out why it's not connecting (@MartensCedric or @PeterGhimself know more about it than I).
Finally, could you explain why docker compose is necessary in this situation? It seems to me like it makes the process more complicated with no added benefit.

Apart from that, it seems like you are on the right track, and you most likely know more than anyone else in the team about docker :+1:

MewtR, 2020-09-21:
@AFknUsername 
Thanks for the feedback.
1. Yeah you're right, I completely forgot about the sudo thing (my user is in the docker group)
2. The problem is that installation instructions are distro specific. Tbh I have very little experience with ubuntu and apt/apt-get which is why I chose not to include installation instructions.
3. So when you launch it, it straight up doesn't connect to the Websocket? I had that error initially but fixed it in  d5fac5e . Can you give me the error message?
4. So this is a matter of opinion honestly. docker-compose is used to manage several containers and I believe it makes more sense to have separate components of the project in different containers because it makes troubleshooting easier since each container has a more specific purpose. I also think it's easier to use because all you need to know is `docker-compose up` and `docker-compose down` basically. But I mean this point is open for discussion

MartensCedric, 2020-09-21:
ill check it out early this week. To know if things work, try to have 2
shells and 'rosgui' in one and 'startgui' in the other. If you can visit
localhost:5000 it's a good start

MartensCedric, 2020-09-26:
Note for peeps : `sudo apt-get install docker-compose` will install a version that wont work so you gotta follow the steps to get 1.27+

MartensCedric, 2020-09-26:
Sorry for the late response, I forgot about this issue lol. I suggest you make a PR so more people can review it, it does seem to _kinda_ work

installation ran pretty smoothly : Got this in the terminal though  : 
```
rosbridge_1   | RLException: ERROR: unable to contact ROS master at [http://ros-master:11311]
rosbridge_1   | The traceback for the exception was written to the log file
```

FYI on our machine we have this in our bashrc

```
export ROS_MASTER_URI=http://localhost:11311                                                    
export ROS_HOSTNAME=localhost       
```

I can see the GUI. This is good.

When I try to check if stuff work I do `sudo docker exec -it <container> bash` (Not sure if this is ideal btw @MewtR )
I try basic ros commands like `rostopic and rosnode` and it says command not found. Also vim is not installed :cry: 

Everything is in root, should it be? Something like `~/Programming/robotics-prototype` would be consistent with some of our aliases (so we wouldnt need to change the path)
[image]

Another thing that I would like to be able to do is launch this file : 

[image]

MewtR, 2020-09-27:
@MartensCedric 
- So like I said earlier, after the `docker exec -it <container> bash` you need to `source /opt/ros/melodic/setup.bash` if you want to get the regular ros commands (rostopic, rosnode etc)
- I can install vim that's fairly straight forward to do
- So the root stuff is definitely an issue but it makes writing these docker config files easier so that's why I did it for now.
- The error you're getting just means that the process inside the rosbridge container was up before the one inside rosmaster container. docker-compose can guarantee that container A starts before container B with the `depends_on` entry but it can't guarantee that container A is "ready" for container B. But that error should only occur on occasion (can rosbridge connect to rosmaster even if rosbridge was started first?).

MartensCedric, 2020-09-27:
ok sounds good, Ill try again.

MartensCedric, 2020-09-27:
ROS works! I just published to the gui using `rostopic pub /battery_voltage std_msgs/Float32 "data: 15.0"` (Just like the initiation task!)

MartensCedric, 2020-09-27:
@MewtR 
> (can rosbridge connect to rosmaster even if rosbridge was started first?).

the rosbridge launch file should launch `roscore` if its not up which is rosmaster

MartensCedric, 2020-09-27:
just tried again and everything worked. You should open a Pull Request now. So I can use the github tools to request some changes instead of commenting on an issue

MewtR, 2020-09-27:
@MartensCedric 
I was gonna make a PR but tried testing the rostopic command and it didn't work (it used to work for me as well).
I have been trying to troubleshoot for a while but haven't had any luck yet.

MartensCedric, 2020-10-02:
Hi, I just tested again. I did `sudo docker-compose up` (no clue if it updated with your latest commits or not), but everything worked just fine.

MartensCedric, 2020-10-02:
> @MartensCedric
> I was gonna make a PR but tried testing the rostopic command and it didn't work (it used to work for me as well).
> I have been trying to troubleshoot for a while but haven't had any luck yet.

You should make a PR even if it doesnt fully work because more people can try out the PR and test if it works or not and give you feedback, since Im not the only reviewer usually

MewtR, 2020-10-03:
@MartensCedric 
So I created the PR. By the way if you want to be sure changes were taken into account simply add the `--build` flag as in `sudo docker-compose up --build`

MartensCedric, 2020-10-03:
Renamed the issue to descope it from Travis. This issue is just to get the docker setup working

PR #465 (2020-10-31): Docker configuration 418
# Assignee Section

## Description
This PR dockerizes the project, allowing users to run it regardless of their environment.
Using docker-compose, three services (group of containers based on the same docker image) are created:
- One running roscore called `ros-master`
- One running rosbridge called `rosbridge`
- One running the actual project called `prototype`

### Steps for Testing

1. Follow the `README` instructions in the `docker` directory to run the project.
2. Once everything is up and running, make sure you see the GUI at http://localhost:5000
3. Open a termi [... cut]
closed 2020-10-31 by MewtR