space-concordia-robotics/robotics-prototype #422: Start migration to melodic

AFknUsername, 2020-07-05:
This issue will keep track of the progress to migrate to melodic.

- [ ] venv works
- [ ] all pip packages are ok
- [ ] all ros packages are ok
- [ ] everything builds on cmake
- [ ] update travis-ci to build on bionic
...

RealWilliamWells, 2020-07-05:
:eyes: I've been doing all of my dev work on Ubuntu 18:04 and Melodic for ~4 months now, and I have yet to run into any ROS related issues. So for at least running the GUI in local mode, I don't think there should be any issues.

AFknUsername, 2020-07-05:
@RealWilliamWells 
That's really good to hear! I knew the process would be relatively painless. The only thing I was worried about was the ROS packages potentially breaking.

PR #423 (2020-08-30): Migrate melodic 422
# Assignee Section

## Description
This PR fast-tracks the migration to Ubuntu 18.04.
Members who upgrade to bionic simply have to execute `EnvironmentSetup.sh`  to get started.

### Changes
1. Install scripts updated to install melodic for bionic, and refactor the scripts into more logical roles.
2. README updated to reflect changes in setup process.
3. travis-ci config changed to reflect the changes to bionic. 
4. submodules updated.

### Steps for Testing
1. Upgrade to bionic (always preferably a fresh install)
2. **GIT CHECKOUT MIGRATE-MELODIC-422**
2. Follow the steps  [... cut]
closed 2020-08-30 by AFknUsername