space-concordia-robotics/robotics-prototype #452: PDS 2

MartensCedric, 2020-09-04:
@shayrin0 if you could fill a basic description of your task it would be appreciated

shayrin0, 2020-09-05:
The task that I'm working on consists of these parts:
 1- Reading temperatures from temp. sensors
 2- Always listening to Master's commands (Master is the PC)
 3- Enabling/disabling fans and motors
 4- Sending data to the Master if needed 
 5- Using an interrupt (and maybe disabling the systems) in case of emergency
 6- Using Arduino bootloader to upload the code on ATMEGA328
 7- Handling the errors
I have worked on some of these parts so far and I'm going to explain each of them respectively.

shayrin0, 2020-09-05:
*Arduino bootloader:
https://www.arduino.cc/en/Tutorial/ArduinoToBreadboard

*Watchdog Timer explanation:
https://circuits4you.com/2018/01/24/tutorial-on-arduino-watchdog-timer-setup/
https://www.instructables.com/id/The-Arduino-Hang-Guardian-Arduino-Watchdog-Timer-T/
 - If something unexpected happens, we may need to use watchdog timer to reset the system.

*Methods for having a serial communication between Arduino and Master are as follows:
 - From Arduino's side: Just read from the serial port
 - From Master's side: Using pySerial, pyDuino
We can stick to pyserial.

closed 2022-10-31 by VimFreeman