space-concordia-robotics/robotics-prototype #454: Integrate navigation with 3D octomapping occupancy cloud

KoaTheKoala, 2020-09-14:
Mina was able to get navigation working in 2D using various ROS packages + some custom written packages. We need to find a way to integrate this with a 3D SLAM map. Currently we are using the octomap to generate a 3D occupancy grid that implements SLAM.

KoaTheKoala, 2020-09-14:
@Mina-Magdy @vishnurk87 

Hey guys, since you have two pieces of the puzzle for this issue, I have created this for you guys to keep track of what you are working on. Do you feel that this description is accurate? If not, do you mind updating it with more details? @Mina-Magdy  since you have a better understanding of the navigation side of things, please fill in the major steps for this task in the template I created. 

If you guys want to get into a call at some point, let me know I'm down. It doesn't have to be in 2 weeks either ;).

PeterGhimself, 2020-09-14:
New branch created for this issue [combine-navigation-with-octomapping-454](https://github.com/space-concordia-robotics/robotics-prototype/tree/combine-navigation-with-octomapping-454)

MartensCedric, 2020-09-19:
Any updates on putting a clear description on this issue @Mina-Magdy @vishnurk87 ?

vishnurk87, 2020-09-19:
@Mina-Magdy Please add/edit the things we need for complete autonomous navigation so we can get a clear understanding of the path to take.