space-concordia-robotics/robotics-prototype #462: Create Science Software Design Diagrams and documentation

MartensCedric, 2020-09-26:
Prior to start the science project we should have a good understanding of the different components of the system. Some documentation should exist so that the members can implement the code by consulting the documentation. Further, it will be crucial to present these to PDR

MartensCedric, 2020-09-26:
[image]

MartensCedric, 2020-09-26:
@Laogeodritt Does this look accurate to you? Am I missing something important or do I have something wrong? Note that this is for Teensy 4.0 software (Excludes CCD as far as I understand it)

MartensCedric, 2020-09-26:
@PeterGhimself @vashmata

MartensCedric, 2020-09-26:
https://docs.google.com/document/d/1N7tWT132qdKljGeyzkHsWxUH83sbp5s3c5rEwCJ0yC4/edit

MartensCedric, 2020-09-26:
[image]

MartensCedric, 2020-09-26:
After some thought I think I should also include the list of the commands that the system should accept and their role

MartensCedric, 2020-09-26:
> After some thought I think I should also include the list of the commands that the system should accept and their role

List of commands : https://docs.google.com/spreadsheets/d/1bE3h0ZCqPAUhW6Gn6G0fKEoOPdopGTZnmmWK1VuVurI/edit

vashmata, 2020-10-03:
> Physical View v1.0
> 
> [image]

By funnel you mean the funnel motor thing?

Does the carousel include the motor AND the limit switches? Just wondering cause you have the temperature sensors as their own thing in the diagram.

Maybe I missed it in @Laogeodritt 's doc but I don't see much about the fan control. Is it simply always on?

vashmata, 2020-10-03:
> Class Diagram v1.0
> [image]

The pumps wouldn't have PID control based on the current design as we don't have a way to obtain feedback. Sorta similar for the funnel (and the carousel which isn't connected to the PID), but in this case it's cause the servos have built-in pid control that we would be able to access if we use the fancy UART commands and would have no feedback for if we use them in open loop with pwm control.

Again you haven't specified the limit switches in the carousel

vashmata, 2020-10-03:
> > After some thought I think I should also include the list of the commands that the system should accept and their role
> 
> List of commands : https://docs.google.com/spreadsheets/d/1bE3h0ZCqPAUhW6Gn6G0fKEoOPdopGTZnmmWK1VuVurI/edit

How about checking the temperature values (in the 2-way commands)?

Even for the one-way commands, it would be nice to have a way to check if the command succeeded or failed. Either by having these commands two-way as well (responding with a success or error message), or having a command that asks for any error states and will receive a message with different types of errors (e.g. "command makes no sense," "invalid carousel position value," etc.

MartensCedric, 2020-10-03:
> By funnel you mean the funnel motor thing?
>
> Does the carousel include the motor AND the limit switches? Just wondering cause you have the temperature sensors as their own thing in the diagram. [...]

This is the physical view. It's just meant to show how the physical things are related to each other.

You have a good point that I added temp sensors but didnt mention limit switches/motors. I think im just gonna remove temp sensors for this physical view because I just wanna show how the components communicate to each other (the OBC talks with the CCD, Teensy 4.0 and Basestation, while the teensy speaks with the rest)

MartensCedric, 2020-10-03:
> The pumps wouldn't have PID control based on the current design as we don't have a way to obtain feedback. Sorta similar for the funnel (and the carousel which isn't connected to the PID), but in [...]

So for the pump, we're just using a timer and don't care about PID control? So what uses a PID then? Does the fans use one to control temperature? If not ill just remove the whole PidController
> 
> Again you haven't specified the limit switches in the carousel

MartensCedric, 2020-10-03:
> > > After some thought I think I should also include the list of the commands that the system should accept and their role
> > 
> > 
> > List of commands : https://docs.google.com/spreadsheets/d/1bE3h0ZCqPAUhW6Gn6G0fKEoOPdopGTZnmmWK1VuVurI/edit
> 
> How about checking the temperature values (in the 2-way commands)?
Yes true, I'll add it. Thanks!
> Even for the one-way commands, it would be nice to have a way to check if the command succeeded or failed. Either by having these commands two-way as well (responding with a success or error [...]

Yes, I could put every command in the Two-Way category. We should consult with @Laogeodritt though

vashmata, 2020-10-03:
this reminds me of something else, the science teensy could also keep a log of errors and timestamps for us to pull up later. Same with temperature readings or other data. Since it only responds when we poll it we're probably missing lots of info that may be useful

vashmata, 2020-10-03:
> So for the pump, we're just using a timer and don't care about PID control? So what uses a PID then? Does the fans use one to control temperature? If not ill just remove the whole PidController

Not sure about the fans... @Laogeodritt ? But for everything else I don't think it's needed.
> 
> > Again you haven't specified the limit switches in the carousel
> 
> Yes good point

MartensCedric, 2020-10-10:
> this reminds me of something else, the science teensy could also keep a log of errors and timestamps for us to pull up later. Same with temperature readings or other data. Since it only responds when we poll it we're probably missing lots of info that may be useful

Yes, I think this is out of scope of science since it would have to be implemented on every teensy right

MartensCedric, 2020-10-10:
v2.0
[image]

MartensCedric, 2020-10-10:
Deployment diagram v2
The purpose of this is just so you can understand what is connected to what device (obc does CCD stuff) while the rest is on the teensy. Not meant to show what has limit switches or whatever
[image]

vashmata, 2020-10-10:
> Deployment diagram v2
> The purpose of this is just so you can understand what is connected to what device (obc does CCD stuff) while the rest is on the teensy. Not meant to show what has limit switches or whatever
> [image]

the fan's purpose is to cool the laser assembly, up to you if you wanna keep them separate but i would put them next to each other on this diagram

MartensCedric, 2020-10-10:
> the fan's purpose is to cool the laser assembly, up to you if you wanna keep them separate but i would put them next to each other on this diagram

Honestly this diagram is just to show that the OBC controls the CCD and the teensy controls the rest. It's not really meant to show the interaction between the components. Its just so someone doesnt think we are doing CCD stuff in the teensy or using the OBC for the pump.

MartensCedric, 2020-10-10:
What about the class diagram @vashmata ?

vashmata, 2020-10-11:
I don't think I have anything more to say, personally I might want to make it explicit that within the carousel you're interacting with limit switches because it's an electrical interface, but that's on you

PR #469 (2020-11-07): Science design 462
# Assignee Section

## Description
Design and skeleton of teensy science code.

- Design documents in PlantUML code in `docs/`
- Added Skeleton code for C++

Notes : 

1. I added some code to the CommandCenter but it's not within the scope of this PR to make it work. I would also like to test it with unit tests.
2. There's some OOP and a malloc in this c++, in both cases I think the memory usage will be kept low

### Steps for Testing
1. Copy paste the diagram code (science-class.uml and science-deployment.uml) in here to see a pic http://www.plantuml.com/plantuml/uml/SyfFKj [... cut]
closed 2020-11-07 by MartensCedric