space-concordia-robotics/robotics-prototype #468: Odroid takes forever to boot sometimes

MartensCedric, 2020-10-03:
[image]

Sometimes the odroid takes more than 15 minutes to be able to SSH in.

PeterGhimself, 2020-10-04:
Seriously, 15 min? I've seen it take a while sometimes but never that long... max a few minutes usually. Did you stop after 15 min or did it connect after 15 min?

Maybe if you're waiting this long (and the issue is not with the power) then it might be because you were switching back and forth between `InternetHog` and `RoverOBC` configurations and not selecting "Basestation Direct" fast enough?

When switching back from internet share mode you need to reboot the odroid and make sure to select "Basestation Direct" on the basestation as the odroid boots and as soon as the system lets you select an ethernet configuration.

Also I can't see but I'm assuming this was `ping odroid` which would resolve to `ping 172.16.1.30`, which wouldn't work if `ROVER_OBC` config wasn't up.

MartensCedric, 2020-10-04:
Me and Koa quit testing because it took more than 15 mins at one occasion
(most of the time at least 150 ping sequences). At the end we couldn't even
connect anymore so we ended up not testing the arm

MartensCedric, 2020-10-04:
Keep in mind this was for just a boot with the Odroid. Internet sharing
worked fine

PeterGhimself, 2020-10-04:
I'm saying _after_ having done internet sharing you need to make sure to in a monkaS fashion that your basestation ethernet profile is "Basestation Direct" _while_ the odroid is booting, otherwise you can run into these types of issues of non-connectivity. I'm assuming that's what the problem was, but that's why I'm not sure what y'all did in the moment.

Do you smell me?

MartensCedric, 2020-10-04:
Kk well I have no clue what profile we were on though. But we hit the kill
switch a few times. Guess we might have been on the wrong profile while
booting but we kept changing it

closed 2022-10-31 by VimFreeman