space-concordia-robotics/robotics-prototype #498: Define a message format

MartensCedric, 2021-01-09:
A definition of the message format should be defined. A google docs should be created (This one can be updated : https://docs.google.com/spreadsheets/u/1/d/1bE3h0ZCqPAUhW6Gn6G0fKEoOPdopGTZnmmWK1VuVurI/edit?usp=drive_web&ouid=111054195812684486479)

The message format should tell us exactly what information is contained, the sheet should make sure that we have the information of the command ID of each command, their description, arguments, argument type etc. So that other members can easily refer to that sheet.

ginywiny, 2021-01-16:
Created a very basic draft just to make a simple idea. Would need others to give their thoughts.

VimFreeman, 2021-01-16:
I added some comments
We should meet anyways today at some point

ginywiny, 2021-01-17:
Created the Rover and Science formats with @VimFreeman and @MartensCedric. PDS and arm commands message format have yet to be done.

PR #510 (2021-01-17): [#498] Implement message format
# Assignee Section

## Description
Added message format

### Steps for Testing
Check if GitHub CI runs

closes #498 

The approval from all software team leads is necessary before merging.

# Reviewer Section

Aside from local testing and the General Integration Test it is implied that static analysis should be included in the verification process.

- [ ] Local Test Performed Successfully
- [ ] [General Integration Test](https://docs.google.com/document/d/1ug0CpA1cIzURP8DDFSvCt2CEJJSwJ6Ta6B1LG_hYk6I/edit) Performed Successfully

For Pull Requests that do not inc [... cut]
closed 2021-01-17 by VimFreeman