space-concordia-robotics/robotics-prototype #513: NIR Software Ubuntu

MartensCedric, 2021-01-19:
https://www.ti.com/tool/DLPNIRNANOEVM

To close this issue, this software should work on Ubuntu

KoaTheKoala, 2021-01-19:
Found this link that might be useful: https://e2e.ti.com/support/dlp/f/94/p/847597/3141022?tisearch=e2e-sitesearch&keymatch=DLPNIRNANOEVM%20linux#3141022

MartensCedric, 2021-01-19:
You might be able to avoid running the software, make a C++ ROS node and use what is described  https://www.ti.com/tool/DLPNIRNANOEVM#software

VimFreeman, 2022-03-09:
@steamypotato Do we have a working way of interfacing with the NIR over the Jetson?

steamypotato, 2022-03-10:
@VimFreeman the NIR was done in C++ , so either we created a C++ ros node, or do some weird python-c wrapping thing that some people do. This was only tested on my PC not the jetson

VimFreeman, 2022-03-10:
A C++ is perfectly fine. We can get someone to test it if the code is complete

closed 2022-10-31 by VimFreeman