space-concordia-robotics/robotics-prototype #545: Create custom mappings for joystick controls.

VimFreeman, 2022-01-30:
We would like to be able to have custom joystick mappings for different people. 
We would also like different mappings for arm and wheel controls and to be able to switch between both using a button.

RealWilliamWells, 2022-01-31:
### To add here are the overall requirements we want for a ROS node that handles all of the mapping:
* All of the mappings should be stored in a YAML configuration file which will then be used for the ROS node to define what actions each button or axis of the rover will do.
* Under each mapping in the config file there should be:
    * The button or axis to assign the command to.
    * The command itself.
    * A list of the parameters for the command.
        * There should be some sort of special systax used to indicate if the parameter must be  based on an axis value. ( This is import for if you want to move say the arm or something at a speed that is dependant on an analog value that would come from the position of a joystick on the controller)
        * For example the string "${axisValue}" as an argument would make it so that the current value of the axis being used is published.
    * Publish rate. (rate to send command at while button is being held or axis is being used)

* Ability to switch between mutiple control configurations within the same file with a button binding on the controller.

### What is currently implemented at this time:
* Partially implemented ROS node that publishes commands upon configurated buttons being pressed.
* Only buttons are implemented with no support for axes.
* Commands do not repeatedly send with buttons being held.
* WIP YAML config file.

The current working branch is [joy-node-0](https://github.com/space-concordia-robotics/robotics-prototype/tree/joy-node-0). You can find the current node for controller [here](https://github.com/space-concordia-robotics/robotics-prototype/tree/joy-node-0/robot/rospackages/src/mcu_control) in the "src" and "include" folders.

### Running the node
Unfortunaltly the current commit on the joy-node-0 branch is incomplete and will not work. Altough there is a version that works with only the features listed under mention earlier in the "What is currently implemented at this time" section. It is under this [commit](https://github.com/space-concordia-robotics/robotics-prototype/commit/5f6e34dce17ffc7af799ba513ab15c463770cd5d). I would recommend to run this first before working on anything in order to ensure that your development setup works.

Checkout to older commit with working minimal feature node (make sure to run this on the joy-node-0 branch):
`git checkout 5f6e34dce17ffc7af799ba513ab15c463770cd5d"

In order to run the node you will first need to run joy_node in order to fetch inputs from the controller:
`rosrun joy joy_node`

Then add the controller configurations to the ROS parameter server:
`rosparam load $(rospack find mcu_control)/config/joy_comms_config.yaml`

Then finally run the node with:
`rosrun mcu_control joy_comms_control`

RealWilliamWells, 2022-05-16:
So I tested the controller node with the rover. Great job! It works almost perfectly for doing analog control with the arm and wheels. There are a few issues to list but for the most part everything worked perfectly. We will need to sit down together at some point and test some specifics more to iron things out.
* Seems like wheel controller mappings are still active for arm module mode but wheel module mode seems to work perfectly with arm control mappings not being active.
* Cannot move mutiple motors at once.
* Turns out the PS4 controller treats the dpad as two axis instead of buttons for whatever reason depsite it not being analog. So the playstation controller button presets will need to be updated to account for this.

AymanShe, 2022-05-17:
Nice. Thanks for testing it @RealWilliamWells. I'm happy it passed (mostly). I will do a round of testing and see if I can address the issues you mentioned.

AymanShe, 2022-06-05:
> So I tested the controller node with the rover. Great job! It works almost perfectly for doing analog control with the arm and wheels. There are a few issues to list but for the most part everything [...]

After some tests I reached the following:
* For the first point I couldn't reproduce, but I found that the cycle between layouts continues to switch as long as the button is clicked, which is not user friendly. One can press the button really fast or I can make it change once and only allow it to change again after the button is released and pressed again.
* For not being able to move multiple arm motors at the same time, I figured that this is because of the nature of the command. When a command is sent, it is sent with a value for one of the motors and zero for all the others ("set_motor_speeds 155 0 0 0 0 0"). I believe there are two ways to solve this, either using the node code to group similar commands into a compound one (which I think will make the code very complex) or use a different command that controls only one motor, something like "set_motor_speed 1 150" where the first argument is the motor number and the the second is the value.
* The dpad was added as axes but there was a typo in couple of the variables and it is fixed now.

VimFreeman, 2022-06-07:
Small suggestion. I think the compound command is the better way to go about it. The code wouldn't need to be much more complex (although I haven't looked at your logic yet so correct me if im wrong). You could check all axes -> fill array -> create command string -> publish. 
I hope it makes sense. 

I believe this method would be better than something like "set_motor_speed 1 150" because in this case if you want to move six motors you'd have to send 6 separate messages over ROS and then to the teensy. Whereas, the current format allows to send only 1.

PR #571 (2022-08-28): Joy node 545
# Assignee Section

## Description
Create the ability to change the layout of the joystick controller by changing a configuration file without changing the code.

- Create a node that reads from Joy Node and publishes Arm Commands and Rover Commands(Wheel).
- Node reads from a Yaml configuration file that has the mapping of buttons to commands.

### Steps for Testing
1. Run ros core.
2. Connect the controller via USB.
3. Run Joy Node.
4. Run mcu controller node.
5. Echo the Arm topic.
6. Echo the Rover topic.
7. Press buttons on the controller and watch the arm and rover commands  [... cut]
closed 2022-08-28 by AymanShe