space-concordia-robotics/robotics-prototype #551: Write ROS driver for 3D CAD Mouse

RealWilliamWells, 2022-03-09:
We want to be able to use the 3D CAD mouse for IK control of the arm. The idea is to write a ROS node that interfaces with the 3D Cad mouses we have around the lab and publish their axes and button states as joy_msgs to ensure capability with our current controller node. 

Requirements:
- [ ] Driver that interfaces with 3D mouse and publishes its state as joy_msgs on update.
- [ ] Automatically find connected 3D mouse(s) on any system running Ubuntu.

RealWilliamWells, 2022-03-09:
This is for the 3D CAD Mouse from 3Dconnexion.

VimFreeman, 2022-03-12:
here is a udev rule i wrote for a webcam. I placed the rule in /etc/udev/rules.d/50-logicam.rules
udev processes the rules in a lexical order, that is what the number in front is for. 

```
ACTION=="add"
, ATTRS{idVendor}=="046d"
, ATTRS{idProduct}=="08b6"
, SYMLINK+="logicam"
```

this creates a symlink based on vendor and product id called /dev/logicam. There are many many more attributes that we can use to identify the cameras when there are multiple (like a serial number). This would be extremely useful on the jetson so that we can access the cameras using stuff like /dev/cam_front or /dev/cam_mast.

vexandmore, 2023-09-16:
Should think how we should do this in web gui, since will require code ported over

vexandmore, 2023-11-18:
I've got a ros2 version of the driver on branch cad-mouse-ros2-driver-551

closed 2024-05-25 by OZiad