space-concordia-robotics/robotics-prototype #582: Investigate LIBs software for integration with rover software

RealWilliamWells, 2022-11-01:
There is a pre-exisitng open source LIBS library written in C# here: https://www.g3gg0.de/wordpress/uncategorized/a-homebrew-libs-software-written-in-c/

The goal of this issue will be to look into how this library works and coordinate with the science team for what they need to do with LIBs. In order to document and outline how we can integrate or fork to use on board our rover

PeterGhimself, 2022-11-05:
[image] [#582 Investigate LIBs software for integration with rover software](https://trello.com/c/ZdM2mevq/82-582-investigate-libs-software-for-integration-with-rover-software)

vexandmore, 2023-09-16:
slate

closed 2023-09-16 by vexandmore