space-concordia-robotics/robotics-prototype #586: Bring back bash aliases that were deleted during repo purge

RealWilliamWells, 2022-11-12:
Bruh

PeterGhimself, 2022-11-12:
```
(venv) peter@boop-Latitude-E6420:~/robotics/OLD_SCHOOL/robotics-prototype$ git branch
* (HEAD detached at bd424c3e)
  investigate-libs-integration-582
  master
(venv) peter@boop-Latitude-E6420:~/robotics/OLD_SCHOOL/robotics-prototype$ find . -iname ".bash_aliases"
./robot/basestation/config/.bash_aliases
./robot/rover/config/.bash_aliases
(venv) peter@boop-Latitude-E6420:~/robotics/OLD_SCHOOL/robotics-prototype$ cat ./robot/basestation/config/.bash_aliases 
#!/usr/bin/env bash

BASE="$ROBOTICS_WS/robot/basestation"
ROVER="$ROBOTICS_WS/robot/rover"

# to start the gui
alias updateEnv="bash $BASE/env.sh >| $BASE/static/js/env.js"
alias roverenv=". $ROBOTICS_WS/venv/bin/activate"
alias startgui="updateEnv && roverenv && cd $BASE && python app.py"
# we have to deactivate venv for this launch file to not break, bug to be resolved eventually
alias rosgui="roverenv && deactivate && roslaunch rosbridge_server rosbridge_websocket.launch"
(venv) peter@boop-Latitude-E6420:~/robotics/OLD_SCHOOL/robotics-prototype$ cat ./robot/rover/config/.bash_aliases 
#!/usr/bin/env bash
ROBOTICS_WS="/home/$USER/Programming/robotics-prototype"
BASE="$ROBOTICS_WS/robot/basestation"
ROVER="$ROBOTICS_WS/robot/rover"
ROSPACKAGES="$ROBOTICS_WS/robot/rospackages"
BASH_A="~/.bash_aliases"
NANORC="~/.nanorc"

# general shortcuts
alias ..="cd .."
alias b="cd -"
alias robotics="cd $ROBOTICS_WS"
alias base="cd $BASE"
alias rover="cd $ROVER"
alias arm="cd $ROVER/ArmDriverUnit"
alias wheels="cd $ROVER/MobilePlatform"
alias rostings="cd $ROSPACKAGES"
alias mcunode="cd $ROSPACKAGES/src/mcu_control/scripts"
alias upload="cmake --build $ROVER/build --target "
alias cmake_clean="cmake --build $ROVER/build --target cleanup && rm -rf $ROVER/build"

# edit thyself
alias editalius="nano $BASH_A"
alias alius=". $BASH_A"
alias editnani="nano $NANORC"
alias nani=". $NANORC"
```

PeterGhimself, 2022-11-12:
bruh indeed

PR #597 (2022-12-21): [#586] Bring back files from purge that turned out to be useful after…
# Assignee Section

## Description
Brought back useful files that were deleted in purge and make fixes to environment setup that accounts for changes made from purge. Tests directory was brought back as there were actual unit tests in there that were useful. Build failures for tests should be addressed in a seperate issue rather than removing all of the tests.

### Steps for Testing
Look over environment setup and confirm that all files under there are in proper locations and are present in repo. Including the additions to  [... cut]
closed 2022-12-21 by RealWilliamWells