space-concordia-robotics/robotics-prototype #617: ROS2 ZED2 Driver

RealWilliamWells, 2023-09-23:
For ROS Humble, we want to have the drivers in place to be able to use the sensors for autonomy. 

## Acceptance
- [ ] [ZED2](https://github.com/stereolabs/zed-ros2-wrapper) is able to be setup with the use of scripts and ROS processes to be usable in RViz2. 

This does not need to be functional yet on the OBC, especially that we intend to replace it so efforts are better focused later.

vutung3196, 2023-09-23:
branch out from ros-migration

RealWilliamWells, 2023-09-23:
You can add any dependencies necessary to robot/rospackages/third-party-dependencies.repos

vutung3196, 2023-10-21:
Hey @RealWilliamWells , I can't find the branch rostwo-migration anymore. Can you help me with this?

RealWilliamWells, 2023-10-23:
Hello, the branch is rostwo-migration-594. You can find it here: https://github.com/space-concordia-robotics/robotics-prototype/tree/rostwo-migration-594

Are you about to checkout to it locally? `git checkout rostwo-migration-594`

closed 2024-05-25 by OZiad